pink88/Tuiss2HA #2: Additional Blinds Hit and Miss

thewillwilson, 2023-10-22:
Hi,

First of all, thanks for developing this integration, I was just looking for some assistance to debug an issue, my first blind connects and works without any issue, however my 2nd blind only works 5% of the time. Each room has their own esphome proxy so I don't believe it's signal strength.

Is there any debugging I can do to investigate further?

pink88, 2023-10-22:
Hi Will,

That is frustrating, I've had issues with my own and proxies in the past,
but managed to get it working pretty stabally after some fiddling.

Could you install the Studio Code Server addon from the Home Assistant
addons store (Settings > Addons), once installed, use the blinds, then
within Studio Code Server find the HomeAssistant.log file and copy whatever
message you are seeing in there?

[image: image.png]

Thanks,
Nigel

pink88, 2023-11-05:
Hi Will, Any update on your issues?

thewillwilson, 2023-11-06:
Hi, thanks for the bump, this slipped my mind!

There isn't much in the logs sadly besides the debug enabled from the addon

`2023-11-06 19:46:14.408 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 0
2023-11-06 19:46:14.408 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.409 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 1
2023-11-06 19:46:14.409 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.409 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 2
2023-11-06 19:46:14.409 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 3
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 4
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 5
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 6
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 7
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 8
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 9
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 10
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed: No backend with an available connection slot that can reach address None was found
2023-11-06 19:46:14.410 INFO (MainThread) [custom_components.tuiss2ha.hub] Connection Failed too many times`

thewillwilson, 2023-11-06:
Just to add to this, the 1 blind out of the 3 that is working perfectly seems to connect first try with no issues.

I've got an esp32 proxy in each room just incase it's anything to do with bluetooth signal strength.

2023-11-06 19:49:01.386 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting Connection to blind. Rety count: 0
2023-11-06 19:49:05.450 INFO (MainThread) [custom_components.tuiss2ha.hub] BleakClient Connected
2023-11-06 19:49:05.450 INFO (MainThread) [custom_components.tuiss2ha.hub] Attempting to set position to: 100
2023-11-06 19:49:05.450 INFO (MainThread) [custom_components.tuiss2ha.hub] BleakClient connected state is True
2023-11-06 19:49:05.451 INFO (MainThread) [custom_components.tuiss2ha.hub] Sending the command
2023-11-06 19:49:05.542 INFO (MainThread) [custom_components.tuiss2ha.hub] BleakClient Disconnected

pink88, 2023-11-08:
Thanks, I did see this before with my own setup and the only surefire way
to fix was when an esphome update triggered them to restart. It would then
be fine for a a week or so before it started again. I'll push an update out
in the next couple of days which handles errors better. Hopefully that will
resolve this for you as it has for me in 99% of cases.

pink88, 2023-11-15:
@thewillwilson any improvement since version 1.1.0?

IainPHay, 2023-11-18:
FYI I have 4 blinds running using this excellent integration now and it's rock solid for me. 3 are set in one group and works well

thewillwilson, 2023-11-23:
Updated last week to it and it's been rock solid, thanks so much for the update!

Let me know if you need any testers for any beta features in the future.

closed 2023-11-23 by thewillwilson