pink88/Tuiss2HA #3: Cant enter MAC address

mattc2, 2023-11-05:
I echo the thanks for developing this intergration.

I dont get an option to enter the MAC address on download and install, just the choice of 'initial' and 'main'. Am I missing something (probably!) 

Thanks

pink88, 2023-11-05:
Hi,

Did you download through HACS? If so it should prompt when you add the
integration with a window like this. Can you share a screenshot of exactly
what you see, so I can try to replicate?

[image: image.png]

Thanks,
Nigel

mattc2, 2023-11-05:
Hi

Your screen shot didnt resolve for some reason, howver this is what I am seeing

[image]

mattc2, 2023-11-05:
FYI I don't get any further dialogue following clicking of DOWNLOAD

pink88, 2023-11-05:
Right okay.

1. Click download.
2. When it's complete you should be prompted to restart home assistant on
the HACs main screen.
3. Restart.
4. When it's restarted go to settings > Devices and Services.
5. Then add new integration.
6. Search Tuiss and then you should get a popup for adding in MAC address
and your name for the device.

mattc2, 2023-11-05:
OK, said it could be me! Sorry, yes confused by HACS intergration download 1st, then a standard HA intergration.  Or to be more to the point a HACS repository 1st. I'm still learning. 

Thanks for your help, Matthew

pink88, 2023-11-05:
No worries. I managed the exact same thing my first time doing this.
Hopefully it works for you, but if you need any help just let me know.

mattc2, 2023-11-05:
Yep, works fine. Thanks again. Your work is apreciated.

closed 2023-11-05 by mattc2