pink88/Tuiss2HA #4: Error working

ellsa100, 2023-11-27:
I added my blinds but when I go to click up or down  on the entity I get the following error:
"Failed to call service cover/open_cover. 'NoneType object has no attribute 'details'"

pink88, 2023-11-28:
Hi Ellsa100,

Can you restart home assistant and confirm it's still an issue for you?

If it is, could you provide some more information please e.g.
- version of Home Assistant
- version of esphome (if using)
- any logs

Thanks
Nigel

ellsa100, 2023-11-28:
Hi Nigel

Thanks.

I did restart HA and issue still happening.

Version of Home Assistant -
Core 2023.11.3
Supervisor 2023.11.3
Operating System 11.1
Frontend 20231030.2

If helpful, Bluetooth in Home Assistant I use is -
BCM20702A0 (0a5c:21f1) by Broadcom
Firmware: home assistant
Hardware: usb:v1D6Bp0246d0546

I don’t use ESPhome - should I? Would this help have my Bluetooth adapter
talk to the Tuiss device?

What is weird is I also tried putting in a random Bluetooth MAC address
into the Tuiss Smartview integration and it lets me add a device even
though it clearly doesn’t exist. I wonder if the Bluetooth in my home
assistant needs some tweaking for it to see my Tuiss blinds?

Thanks very much

ellsa100, 2023-11-28:
Also I am using my thin client’s in built Bluetooth module if that is
relevant?

pink88, 2023-11-29:
I only mention esphome because that is what I use and have tested with, but it should really matter, as home assistant handles the bluetooth adapter side of things itself.

Regarding the MAC address working when it doesnt exist, that is just lazy coding on my part as I havent added any validation.

Would you be able to pull the logs from when you get this error. Easiest way is to go Settings > System > Logs then click the log entry related to this error and copy all the info it then gives you.

ellsa100, 2023-11-30:
I tried adding the Shelly Bluetooth proxy too. I looked at the diagnostics log for both built-in Bluetooth module and the Shelly Bluetooth proxy and, amongst the bluetooth devices it can 'see', the Tuiss bluetooth. Not sure why it is not (I think) getting accurately picked up by this integration? I can send the bluetooth diagnostics logs if helpful?

pink88, 2023-11-30:
If you can share the logs I can take a look. And you're saying that there are no logs generated when you click open/close and get the "Failed to call service cover/open_cover. 'NoneType object has no attribute 'details'" error. I would have expected a recent log to be visible after trying that action and then you can copy across the whole thing.

This integration works by connecting to the blind each time and then sending a command, so its not actively listening to the blinds or anything like that because they arent that clever, so i would expect both your inbuilt and the shelly bluetooth modules to be able to do that without much difficulty.

My wife did tell me that she saw the same error you mentioned, but it resolved on a reboot. 

If you can send logs, I have some time to look into this tomorrow.

ellsa100, 2023-12-01:
Hi - yes, here is the log:

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:230
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 30 November 2023 at 1:40:19 pm (16 occurrences)
Last logged: 11:23:18 am

[140337789227712] 'NoneType' object has no attribute 'details'
[140337421242688] 'NoneType' object has no attribute 'details'
[140336487241536] 'NoneType' object has no attribute 'details'
[140337556519360] 'NoneType' object has no attribute 'details'
[140337417710016] 'NoneType' object has no attribute 'details'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/cover.py", line 163, in async_set_cover_position
    await self._roller.attempt_connection()
  File "/config/custom_components/tuiss2ha/hub.py", line 80, in attempt_connection
    await self.blind_connect()
  File "/config/custom_components/tuiss2ha/hub.py", line 90, in blind_connect
    client: BleakClientWithServiceCache = await establish_connection(
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak_retry_connector/__init__.py", line 330, in establish_connection
    if IS_LINUX and (devices := await get_connected_devices(device)):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak_retry_connector/bluez.py", line 425, in get_connected_devices
    if not isinstance(device.details, dict) or "path" not in device.details:
                      ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'details'

pink88, 2023-12-02:
Can you try something for me please, because this error implies that the device doesnt exist, which I believe at the point in my code all I'm really doing is passing the mac address of the blind to home assistant to connect to. 

Can you check your shelly bluetooth proxy is in active mode
Install StudioCodeServer addon from the home assistant add on store
Reinstall the tuiss2ha add on and readd your blind
Let me know the results

ellsa100, 2023-12-02:
Hi 

So firstly, I don't have any other bluetooth devices connected to HA, so I don't know...not sure if the connections are not working properly?!?

I turned Shelly bluetooth proxy into active mode and followed your steps. Below is the error. Thanks again!

2023-12-02 17:34:00.787 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139668235400000] 'NoneType' object has no attribute 'details'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/cover.py", line 163, in async_set_cover_position
    await self._roller.attempt_connection()
  File "/config/custom_components/tuiss2ha/hub.py", line 80, in attempt_connection
    await self.blind_connect()
  File "/config/custom_components/tuiss2ha/hub.py", line 90, in blind_connect
    client: BleakClientWithServiceCache = await establish_connection(
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak_retry_connector/__init__.py", line 330, in establish_connection
    if IS_LINUX and (devices := await get_connected_devices(device)):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak_retry_connector/bluez.py", line 425, in get_connected_devices
    if not isinstance(device.details, dict) or "path" not in device.details:
                      ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'details'

pink88, 2023-12-02:
Okay, so the issue appears to be that either your integrated bluetooth module and/or the Shelly arent able to detect the blind, which is intetresting as you said that your Shelly was reporting it as available. The 3rd line in the first set of logs you sent should say the MAC Address of the blind instead of None, which is theyn causing issues downstream. 

I think the best next steps are probably to troubleshoot the shelly proxy on your end OR I can recommend just grabbing an esp32 board for about £5, configuring using this guide and placing near to your blinds:
https://esphome.io/components/bluetooth_proxy.html

Not sure what else I can do I'm afraid. Would be curious to here if you manage to get anywhere though to see if there is anything I can do. I'll atleast add some error handling in for when devices cant be reached like this.

IainPHay, 2023-12-02:
Get the M5 stack as they are small, well built and effective. The. You can use https://esphome.io/projects/index.html to generate the proxy on it really easily. Well you can use that for the bare boards as well but the M5 is just a best unit for not much more cost.

ellsa100, 2023-12-02:
I enabled debug logging on Shelly and looked at log. I saw the below. I have put in bold the MAC of the blinds....so it looks like the Shelly bluetooth proxy is seeing the blinds somehow?!?!

> 2023-12-02 22:31:27.352 DEBUG (MainThread) [aioshelly.rpc_device.wsrpc] recv(10.0.0.18): {'src': 'shellyplus1-b8d61a865414', 'dst': 'aios-140212699251600', 'method': 'NotifyEvent', 'params': {'ts': 1701556285.61, 'events': [{'component': 'script:1', 'id': 1, 'event': 'ble.scan_result', 'data': [1, '**e3:09:5d:57:bf:2b**', -94, 'AgEGDf9ZAOMJXVe/KxEjBQEHCVRTMzAwMA==', ''], 'ts': 1701556285.61}]}}
> 2023-12-02 22:31:27.353 DEBUG (MainThread) [aioshelly.rpc_device.wsrpc] Notification: NotifyEvent {'ts': 1701556285.61, 'events': [{'component': 'script:1', 'id': 1, 'event': 'ble.scan_result', 'data': [1, '**e3:09:5d:57:bf:2b**', -94, 'AgEGDf9ZAOMJXVe/KxEjBQEHCVRTMzAwMA==', ''], 'ts': 1701556285.61}]}

ellsa100, 2023-12-02:
> roxy on it really easily. Well you can use that for the bare boards as well but the M5 is just a great unit for not much more cost.

thanks!

pink88, 2023-12-03:
I've created a beta release which _**might**_ fix your issue. Could you try it out? It might show by default, depending on your selection, otherwise you can go into HACS, click the integration, click the ... in top right, redownload, select show beta and then pick 1.1.1-beta.
Let me know if anything changes.

ellsa100, 2023-12-14:
Hi - I bought the M5 Atom and set up ESPHome on it....AND IT NOW WORKS WITH MY TUISS BLIND!!!! So seems like there was something wrong with the Shelly Bluetooth Proxy. Thank you!!!

pink88, 2023-12-17:
Glad its working for you now. I might pick up a shelly just for my own knowledge of why it didnt work. Will close this ticket now though.

closed 2023-12-31 by pink88