pink88/Tuiss2HA #6: Blind state

kc-nucleon, 2024-01-09:
Hello, I notice the blind state is set by the code, I'd like to expand out the application so it can ask the blind it's position as in the native app do you happen to know the hex code to send to blind to get this? I ask as we have a remote for this but it causes HA to lose the state of the blind?

pink88, 2024-01-12:
Hi,

Unfortunately the blinds don't advertise their state, so there is no way to
get that information without connecting first. The only way around this
would be by running a service to a schedule which connects to the blind and
retrieves the current position. I have a new version I'm testing that does
this for battery status, so could implement a similar thing for position. I
wouldn't recommend against running this sort of service in real time
though, as it will probably stop your remote from being able to connect and
may drain battery quickly.

Thanks
Nigel

kc-nucleon, 2024-01-16:
Yea, I didn't think they would have a state as such but they would have have position and you set top and bottom, I assume this is how the Tuiss app does it.  And yea, it's the kind of thing you you would only ask the the blinds about every 3 hours or so, depending how important the information is to you, do you happen to know the hex codes for current position, i might be able to expand it out?

pink88, 2024-01-19:
I'll have a look as well, but this is the hex code which should get the
blinds position as a response.
ff78ea41d10301

 If it's the same encoding as the set the position, then there is some
logic to convert the response into the position value. To set the position
I use this hexcode, which converts a % into some hex values and sends to
the blind. Imagine it will need reversing to get the position.:
"""Convert the blind position."""
        callStr = "ff78ea41bf03"
        outHex = round((((100 - userPercent) * 10) % 256), 1)
        if outHex == 256:
            outHex = 0
        if userPercent > 74.4:
            groupStr = "00"
        elif userPercent > 48.8:
            groupStr = "01"
        elif userPercent > 23.2:
            groupStr = "02"
        elif userPercent >= 0:
            groupStr = "03"
        hexVal = str(format(int(outHex), "#04x"))

pink88, 2024-01-26:
Btw. I have released a beta that includes a service for this, alongside some other changes.

unregist, 2024-01-29:
I ran into issues with that beta due to issues with the bytearray/split_data function
I've made some modifications to your code in pull request #8 to fix this.
Now working for my blinds

pink88, 2024-02-19:
Thanks, this is now included in version 1.2.0. Closing this issue.

closed 2024-02-19 by pink88