pink88/Tuiss2HA #10: Could we add a "cover.stop_cover" action?

ThinkerDreamer, 2024-02-12:
Hi! This is less an issue and more a question/feature request. I have this integration working on my Tuiss blinds and I love it. I have widgets on my phone to make the blinds go up and down, and one to call the [`cover.stop_cover` service](https://www.home-assistant.io/integrations/cover/), but it looks like the stop function isn't implemented here yet. Is that correct? 

I am happy to work on this myself if someone could point me in the right direction. I've written some Python before, but never for Home Assistant. I usually work in Rust and TypeScript. 

Basically what I would like to do is make the blinds go up or down a little and then stop them when they get to the right height. Has anyone been able to make that work? Do you have any advice about what to try or do?

Thanks again for this excellent integration! 🎉 🙌🏻

ThinkerDreamer, 2024-02-12:
I've looked at `custom_components/tuiss2ha/hub.py` and `custom_components/tuiss2ha/cover.py` and I think I understand the basic idea. I'll see if I can find another `cover` entity(?) in Home Assistant to see how they implement the `cover.stop_cover` service.

pink88, 2024-02-13:
It can implemented pretty easily and I had it working before. The issue I
found was that I could click stop and half the time the command wouldn't
send in time and so I removed it as it never really served it's purpose.

I will take another look now I understand the blinds a little more and see
if I can make it more reactive.

ThinkerDreamer, 2024-02-13:
Hi! Thanks! That would be great! Let me know if I can do anything to help.

pink88, 2024-02-19:
There is a new beta out 1.3.0-bneta1, which you can get through HACS or here if pre-releases are selected. This includes the STOP command and a bit of rework to get everything working properly. There are a couple issues I have spotted that I will clean up over the next couple of days, but it should work. Give it a go and let me know if its working for you.

ThinkerDreamer, 2024-02-20:
Thank you! I'll check it out now!

pink88, 2024-02-28:
I've pushed the update into the main branch and released it. Will close this issue. If you spot anything else, please open a new one.

closed 2024-02-28 by pink88