pink88/Tuiss2HA #11: With which Tuiss motor is this working?

atomass, 2024-02-27:
Hi @pink88,
thank you very much for the time and work you've put into this.
Unfortunately, your code is not working with my Tuiss motor. So, I did some "test" and I found out that my Tuiss motor expects a password at the beginning of the connection. Have you found anything like that?
I'm working on my own code and I'll share it when it will be ready to be tested. I'm planning on supporting multiple (all) Tuiss motors.

pink88, 2024-02-27:
Hi Andrea,

Interesting. No I've not, I've only tested on TS3000 devices, which were
roman blinds from blinds2go, but they just required a basic bluetooth
connection without auth. Where did you get yours and do you have a  model
number?

Thanks,
Nigel

atomass, 2024-02-27:
Hi, thank you for the quick reply.
Mine is a TS2900 device, I bought it recently from a Spain reseller. Maybe they introduced the password on a firmware upgrade?
Are you using their Tuiss app to interact with the blind? Or do you have another app?

pink88, 2024-02-27:
Yes, use the Tuiss app. Interestingly there was an issue before where a
change was submitted to handle ASCII characters and was tested against a
TS3000 and TS2900 per their comments, so at least for that user it was
working in the past.

Let me know if I can help while you are debugging and if you can make sure
you've forked from the dev branch as it contains some changes to how I
connect and disconnect from the blinds.

unregist, 2024-03-04:
Hi. I have the TS2900 (Australian reseller).
I've not got any issues (version 1.3.1 currently). No request for a password from the blind.
Tuiss App works fine as well as this integration, again no password request.
Let me know if you need anything from my blind to troubleshoot.

closed 2024-07-17 by pink88