pink88/Tuiss2HA #14: Missing bleak-retry-connector  dependency

gpbenton, 2024-02-29:
I have just tried to  update from 1.2.0 to 1.3.0 with HACS, but I get this error 
`2024-02-29 16:37:30.288 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'tuiss2ha': Requirements for tuiss2ha not found: ['bleak-retry-connector>=3.4.0'].
`
Is this something HACS should download, or is it a manual step?  This is my only use of HACS, so I'm not sure how it is supposed to work.

pink88, 2024-02-29:
Hi, I'm not sure why it is failing as it wasn't failing on my HA instance. In any case, I have pushed out 1.3.1, which reverts back to an older version of the bleak-retry-connector. Please let me know if this works.

IainPHay, 2024-02-29:
FYI it works perfectly on mine as well after the 1.3.0 it's a very nice update 👍

gpbenton, 2024-03-01:
Thanks,  1.3.1 works fine for me.  But I am not sure that is the correct solution, since 1.3.0 works for others.

I think I need to establish where this library is supposed to come from.  I had thought that HACS was a sort of package manager, responsible for downloading dependencies, but looking more closely I can't see where it would do that.  

From that I am beginning to think that its probably dependent on my HA setup.  I am running a docker installation, so would that make a difference?

pink88, 2024-03-01:
Agreed, but there is no hard dependency on that version, so rollback is
okay for now. It should be managed my home assistant I believe as a part of
the install, rather than HACS.

Out of curiosity, how have you installed your Home Assistant instance, I'm
wondering if there is a difference between Docker or HAOS on a pi, for
example.

gpbenton, 2024-03-01:
If it is dependent on the HA version of the library, then that probably explains it.  The HA test version I was running was 2023.12 .

If your custom component has no dependency on the later version of the library, its probably best to specify the earlier version to maintain compatibility with older versions of HA.  Unless its incompatible for other reasons of course.

But maintaining compatibility across different versions of HA will be a major ongoing maintenance problem, and I can understand if you don't want to bother with that :)

pink88, 2024-03-09:
This was my bad, I didnt consider that's how it works and so was only bumping up versions because they were available. Will consider this going forwards or alert on breaking changes.

Closing this ticket now.

closed 2024-03-09 by pink88