pink88/Tuiss2HA #16: Any chance of adding Shelly Plus 1 Bluetooth Proxy to Tuiss2HA ?

MarkRocketman, 2024-03-03:
I am running Tuiss2HA on current version of HASSIO via a Pi3b. The signal strength is just too weak to reach my Blind2go Tuiss. I have a Shelly Plus1 in the same room controlling lights. I would like to use it as a Bluetooth Proxy but do not have the skills or knowledge to configure it is there any way to integrate a proxy?   See my fix to date a Hardware HAC  Not nice but it works ! 
[image]

pink88, 2024-03-03:
I've not got any of these devices, but the integration isn't device
specific, it just uses Home Assistances own Bluetooth libraries.

Looking at the official documentation and a couple forums though it seems
that the shelly devices aren't supported when it comes to sending commands
and can only be used to listen to Bluetooth advertisements. See the shelly
section here:
https://www.home-assistant.io/integrations/bluetooth/

To range exchange for controlling blinds it seems the only option at the
moment is an ESP board, so I don't think there is anything I can do I'm
afraid.

Btw love the hardware hack.

MarkRocketman, 2024-03-03:
Thank you for the feed back.  I was  conning  the conclusion that it was just listening so half a proxy 😀 Your software integration works well  I am just short on signal.Thanks again,Mark

https://www.home-assistant.io/integrations/bluetooth/

closed 2024-03-09 by pink88