pink88/Tuiss2HA #17: 'None_Type' object has no attribute

r3g001, 2024-05-04:
After installing ESP Home and Tuis2HA, I add a new integration, add the mac address of the blind and give it a name, but then after that I can't get the blind to respond.

Only the "Open" and "Stop" icons are available and the "Close icon" is greyed out.

When clicking on Stop, the status bar shows the message;
"Failed to call service cover/stop cover. 'None_Type' object has no attribute 'is_connected'".

When clicking on Open, the status bar shows the message;
"Failed to call service cover/stop cover. 'None_Type' object has no attribute 'details'".

HA log contains;

AttributeError: 'NoneType' object has no attribute 'state'
2024-05-04 18:23:53.443 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547546259648] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2738, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2779, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1047, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/cover.py", line 189, in async_stop_cover
    await self._blind.stop()
  File "/config/custom_components/tuiss2ha/hub.py", line 188, in stop
    _LOGGER.debug("%s: Stop failed. %s", self.name, self._client.is_connected)
                                                    ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'is_connected'
2024-05-04 18:24:04.887 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547546259648] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2738, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2779, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1047, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/cover.py", line 146, in async_open_cover
    await self._blind.attempt_connection()
  File "/config/custom_components/tuiss2ha/hub.py", line 116, in attempt_connection
    await self.blind_connect()
  File "/config/custom_components/tuiss2ha/hub.py", line 125, in blind_connect
    client: BleakClientWithServiceCache = await establish_connection(
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 343, in establish_connection
    if IS_LINUX and (devices := await get_connected_devices(device)):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/bluez.py", line 428, in get_connected_devices
    if not isinstance(device.details, dict) or "path" not in device.details:
                      ^^^^^^^^^^^^^^

Tuis2HA is v1.3.1

Am I doing something wrong or is there a problem with the integrations?

r3g001, 2024-05-05:
Hi Nigel,

Thanks for getting back to me.

For further background info, I'm running a Home Assistant Yellow which I understand does not have built in bluetooth, but I read from an ESP Home tutorial that adding the Bluetooth proxy enables bluetooth in the same as as if I added a bluetooth dongle to the Yellow. I'm quite new to HA, so I'm working with lots of new things all at the same time which is never good when it comes to fault finding. I have confirmed that my ESP32 board is connected in HA and I can read properties of the entity in a dashboard like connected SSID, but I don't know how to confirm if bluetooth is working.

I followed your instructions to turn on debug logging and tried to activate the service to return the blind position, but the service call fails.

Here is the log output.

Thanks

2024-05-04 09:25:01.180 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hikvision_next which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-04 09:25:01.183 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration tuiss2ha which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-04 09:25:01.186 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-04 09:25:06.215 WARNING (MainThread) [hass_nabucasa.cloud_api] Fetched https://alexa-api.nabucasa.com/access_token (400)
2024-05-04 09:25:11.162 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-05-04 09:25:26.863 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.helpers.event. This is deprecated and will stop working in Home Assistant 2024.11, it should be updated to import functions used from event directly at custom_components/hacs/base.py, line 630: self.hass.helpers.event.async_track_time_interval(, please create a bug report at https://github.com/hacs/integration/issues
2024-05-04 13:55:15.293 WARNING (MainThread) [homeassistant.helpers.entity] Update of binary_sensor.hive_hub_status is taking over 10 seconds
2024-05-04 13:55:20.292 WARNING (MainThread) [homeassistant.components.binary_sensor] Updating hive binary_sensor took longer than the scheduled update interval 0:00:15
2024-05-04 18:23:05.967 ERROR (MainThread) [homeassistant.components.binary_sensor] Error adding entity binary_sensor.tuis_lounge_battery for domain binary_sensor with platform tuiss2ha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 892, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1358, in add_to_platform_finish
    await self.async_added_to_hass()
  File "/config/custom_components/tuiss2ha/binary_sensor.py", line 74, in async_added_to_hass
    if last_state.state == "on":
       ^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'state'
2024-05-04 18:23:53.443 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547546259648] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2738, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2779, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1047, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/cover.py", line 189, in async_stop_cover
    await self._blind.stop()
  File "/config/custom_components/tuiss2ha/hub.py", line 188, in stop
    _LOGGER.debug("%s: Stop failed. %s", self.name, self._client.is_connected)
                                                    ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'is_connected'
2024-05-04 18:24:04.887 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547546259648] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
[... cut ...]
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 343, in establish_connection
    if IS_LINUX and (devices := await get_connected_devices(device)):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/bluez.py", line 428, in get_connected_devices
    if not isinstance(device.details, dict) or "path" not in device.details:
                      ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'details'

danmansfield, 2024-05-05:
Do you have an actual Bluetooth Proxy ESP Module like this one: [Atom](https://thepihut.com/products/atom-lite-esp32-development-kit) ?
I literally set this up a few days ago - the module is so easy to do and only costs £9. The wizard adds it to HA and then this integration will find all the blinds you have already added.
I came across this issue because I had the same error today - it turned out that the someone had unplugged the module!

r3g001, 2024-05-05:
Hi Dan,
No the module I'm using is one of these;
https://www.amazon.co.uk/dp/B0C6QHLGJG?psc=1&ref=ppx_yo2ov_dt_b_product_details
I read from another site that it works well with ESP Home, but the OP was not using it for Tuis2HA, so I can't be certain that it is compatible.
I'll give the Atom a try though - only just ordered some stuff from PI Hut yesterday, so gives me an excuse to order some more :-)

pink88, 2024-05-06:
Hi Richard,

I think all the log is really stating is that your bluetooth is unable to
connect and because there is no state for the blind set on the initial
connect, its throwing some errors. I will add some handling for this in an
upgrade.

The ESP32s you linked to are the same type I am using in my home, so it
should be working. When you go to your ESPHOME dashboard, the click Logs >
Wireless Logs for one of the bluetooth proxies are you getting something
like this?
[image: image.png]

My best guess would be that the proxies are not set up correctly.
In the yaml file for the proxy (click edit within ESPHOME), you should have
some config like:
# For Bluetooth Proxy
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
  cache_services: true

Nigel

r3g001, 2024-05-06:
Hi Nigel,

Looks like we have progress :-)

When I checked the config of the ESP32, there was no section for Bluetooth Proxy, so I added in your example from above and restarted everything, then it picked up the first blind that I added. I had to call the blind position service to get it to fully work, but after that it opened and closed fine.

I added a couple more blinds that are in range of the ESP32, but they didn't seem to want to work at first, but after a couple of restarts and calling the position service again they came online.

Interestingly, I flashed the ESP32 with the bluetooth proxy using the web page method from the ESP Home site, so it seems odd that the config was missing the bluetooth section. I actually flashed it a second time when it didn't work to begin with, so there must be something wrong with the bluetooth proxy script.

I've got some more blinds to add, but they are not in range so I need to configure another proxy to do those.

Many thanks for the assistance with this.

Cheers,

Richard

closed 2024-05-06 by r3g001