pink88/Tuiss2HA #18: Error Message when Using Tuiss HA Integration

Adrian281701, 2024-05-13:
Apologies new to HA so still figuring m way through.

I have asmart blind, which uses the Tuiss platform, on the Tuiss app on my phone it works fine. I have the Tuiss integration installed within my HA installation, however I press a button to operate the blind within HA I get an error message 'Failed to call service cover/open_cover. None Type object has no attribute details. 

My seach did lead me to another similar issue you helped another user resolve, however my understaing there was a Bluetooth proxy element to this. I do not believe I have this,, but i am open too being corrected.  My understaing is that my phone connects via WIFI to the blind.

Home Assistant version details are:
Core: 2024.5.3
Supervisor: 2024.05.1
O/S 12.3
Frontend 20240501.1

Integration was installed directly through HACS.

As I said above I am new to this so any help would be appreciated

Thanks

Adrian

IainPHay, 2024-05-13:
Hi Adrian,

I'm not the author of this but I can comment and say they definitely don't connect via WiFi. Tuiss blinds are Bluetooth Low Energy (BLE) so you definitely need to use a Bluetooth proxy or Bluetooth on your HA device if in range and it will work really well.

The https://esphome.io/projects/index.html page will really help with this and the suggested M5 stack atom lite works really well for me. I have one in each room with the blinds in and it's working flawlessly.

Hope this helps

Iain

Adrian281701, 2024-05-14:
Hi Iain
Thanks for your reply, very imformative, I will continue to investigate. I have a Broadlink M4 device, could I use this in the same way? As it is already working in HA
Adrian

pink88, 2024-05-15:
Hi Adrian,

I'm not sure about that device as it appears to be a RF/IR controller and
not bluetooth. As Iain (thank you Iain!) mentioned your best best is to
look into ESPHome to get one or more bluetooth proxies set up to control
the blinds. Once you have these devices, getting them set up and running it
quite easy and then the integration should just start to work with your
blinds.

Thanks,
Nigel

Adrian281701, 2024-05-15:
Thanks Nigel, looking into that, slight problem with ordering at the moment

Adrian

Adrian281701, 2024-05-20:
Hi Nigel
As per Iains suggestion I got a bluetooth proxy,, the M5Stack Atom Lite. I have followed the auto install process., and by some miracle I managed to get it online within HA and working with the blinds. First question I installed the software while it was connected to my laptop, am I able to unplug it, and plug it into the HA server? I have a HA GREEN, not sure if the USB ports carry power.

Second question is regards the integration. When I use the slider control it does not show a percentage of closing, am I missing something, control does however activate the blind.

Thanks for the help so far, and hopefully on the above

Adrian

Adrian281701, 2024-05-21:
Hi 
HELP PLEASE - I think I have messed things up. ESP told me there was updates for the proxy, ran the update. As with the original installation the process seemed to get into a loop, I have attached the logs upto the point i halted the process.
[logs_atom-bluetooth-proxy-31a13c_run.txt](https://github.com/pink88/Tuiss2HA/files/15393960/logs_atom-bluetooth-proxy-31a13c_run.txt)
I stopped it after a time, however the BlueTooth Proxy device in the ESPHome dashboard reports the device as offline, and I get the same original error message. 
Forgive me I am noob at this, but from what I could see spinning through log was lots of scanning for wifi, but not finding anything, and showing an ip address not on my network. NB correct network SSID and Pword appear in the secrets area, and YAML file I can download from the device points to the network setting in the secrets area.
Should I have not updated? Think I can reset it back to the state it was previously which seemed to work, pre update.
Hope you can give me some pointers
Thanks in advance
Adrian

pink88, 2024-05-22:
Hi Adrian,

Great to hear that it was working for you! To answer your questions:

1. Once the M5 has been configured using the laptop and linked to HA it can
be plugged into any usb port/outlet as long as it's on the WiFi network it
will work from wherever.

2. These blinds don't provide a in sync position update as they open or
close, so while they moving to position you just get "opening" or "closing"
status.

3. I've never had an issue with doing an update to any esphome devices. The
process I've always followed has been by installing the Bluetooth Proxy
project from here https://esphome.io/projects/, then adopt it from the
esphome dashboard in home assistant. If you are getting issues, best to
reach out on either the esphome GitHub or the home assistant forums.

Thanks,
Nigel

Adrian281701, 2024-05-23:
Hi Nigel

Thanks for responding, appreciate you taking the time. Fgured point 1 out by simply unplugging from the laptop and plugging into the HA server hadware, easy peasy....lol

I figured as much on point 2, but it got me thinking. When I am using the slider I dont need an in-sync position status from the blind controller, surely I just need to read the position of the slider is moved too as that is the position I am setting and sending to the blind? or am I missunderstanding (highly possible).

I did manage to reset the BT proxy back to a working state, by doing as you mentioned, device update message stopped, so guessing is also forced an update. I am assuming the update HA is trying to install is not quite the same as the one deployed from https://esphome.io/projects/, I will follow your lead and do the same for future updates. However HA is still telling me the device needs a firmware update (from 2024-5-0 to 2024-5-2) guessing this is not part of the BT Proxy deployment. Anyways will reach out on ESPHOME github and investigate further.

Bottom line I have my blind working through HA so thanks for the assist.

One last question before I go, how many devices (blinds) can I control through one BT proxy? or do i need one for each?

Adrian

pink88, 2024-05-24:
Great to hear.

I let Home Assistant do the updates to my esp devices without any issue ,
so you likely just hit a bug or failed update that "broke" the device.

In theory you could connect more blinds than you'd ever need to home
assistant and it will manage the connections to the blinds via the proxies.
Each proxy can handle 3 simultaneously connected devices, but 1 proxy can
communicate to more, just only control 3 at the same time. What you'll find
is the main limitation is the distance between the proxy and the blind. I
tend to have 1 proxy within 4 meters of a device and add a new one if the
signal would otherwise be going through a couple walls or through a ceiling.

Bit of trial and error.

closed 2024-07-17 by pink88