pink88/Tuiss2HA #19: Thanks and help

Carphone12, 2024-05-27:
Hi,

Just stumbled on this - i have 2 blinds for blinds2go and have been using a BroadLink pro to control them by learning the RF 433mhz remote buttons. This works OK about 80% of the time so found this option when doing some googling.

I have a HA yellow box with a built in BT adapter into the board. When i try and control the blinds using this great integration it is very slow to action the command after pushing the down arrow (approx 4 secs). And if i keep hitting the buttons / not be patient I get one of the two errors below.

Do you think adding a Bluetooth proxy will improve the speed? What is the typical log from pressing to action of a fully working implementation?

I had know idea these were both Bluetooth and RF devices / i can communicate with them using both .. 

Error 1
Failed to call service cover/close_cover. [org.bluez.Error.Failed] Not connected

Error2
Failed to call service cover/open_cover. C5:5C:35:C2:xxx - C5:5C:35:C2:xxx: Failed to connect after 11 attempts): No backend with an available connection slot that can reach address C5:5C:35:2:xxx was found:
The proxy/adapter is out of connection slots or the device is no longer reachable; Add additional proxies

Carphone12, 2024-05-27:
PS - big thanks for building this integration - amazing work

pink88, 2024-05-27:
Hi,

Unfortunately my experience of these blinds is that there is a delay
connecting to the blinds of 4-20 seconds. A proxy may improve things if
it's physically closer, but it will rarely be instantaneous.

This seems to be a limitation of the BLE capability of the blinds
themselves as I get the same experience controlling them from the Tuiss app
and the integration is just sending the same sets of commands as that.

It's interesting that you have been able to control them from the broad
link though, as I was unaware they worked with RF, do you know which model
of blind you have?

Thanks,
Nigel

Carphone12, 2024-05-27:
[image]

Pic of the motor in my blinds attached - maybe the model number will help you.

Good point on the app connection time. I have ordered a Bluetooth proxy to try and see how it compares. Will let you know how it goes.

The rf remote controls that can be ordered with these blinds work fine with my BroadLink. I real faff to get home assistant to learn them as have to try a lot of times. I have also figured out today that the channel you use the remotes on impacts how reliable they are. I had my remotes on channel 1 which meant the button presses only worked c.70% of the time. This mean my BroadLink was also not working all the time. I changed the remotes to ch14 and got HA to re-learn the RF codes on this channel and now the remotes and HA seem to be 95% reliable.

I have actually got an Alexa routine to trigger my HA script so I can control them by voice - it works very quickly.

jb123567, 2024-07-20:
I use Bluetooth proxies in ESPHome and the blinds generally react within 1 - 2 seconds. I use Ethernet ESP32 that are powered via POE. I know that there are problems with using WiFi based ESP32 for bluetooth proxies.

Carphone12, 2024-07-21:
Thanks - My BT proxy is setup and working. I can now control these blinds through this BT integration as well as through the RF repeater (Broadlink). Pros and cons for both - but the RF approach gives instant response times that i really like.

closed 2024-10-05 by pink88