pink88/Tuiss2HA #20: Failed to call service cover/open_cover. 'NoneType' object has no attribute 'details

ValiantGB, 2024-07-03:
I have two TS3000 Tuiss blinds and have connected these successfully on my mobile phone using the Tuiss app.  The blinds both work fine using the app.

I have the Tuiss integration installed within my HA installation (HACS), however when pressing any blind icon on my dashboard, I get the above error message.

I have an Atom M5Stack flashed with ESPHome BLE Proxy, adopted via the ESPHome HA integration and this is located in the room with the blinds.  Communication / Logs between HA and the BLE Proxy appear normal and work fine.

Is there a configuration step in the set up that I am missing?  How do you "connect" the Atom M5Stack Proxy to the blinds so that it can parse the HA instructions to the blinds and vice versa, which I assume is the problem?

Many thanks

pink88, 2024-07-04:
Hi,

Your set up sounds like it should work, but this error usually means that
either the blind MAC address is wrong or there is an issue with the ble
proxy.

   1. Try re-adding the blind
   2. If that doesn't work, enable debug logging for the integration, try
   to control the blind again and then share the logs. Need the logs related
   to tuiss and esphome, you can find both under settings > system > logs,
   then search tuiss for tuiss logs and for the esphome logs from the same
   screen, change the top right dropdown to "esphome" from "home assistant
   core".

Thanks

ValiantGB, 2024-07-04:
All working now - I think the problem arose because I did the Tuiss integration _before_ I did the ESPHome BLE Proxy - I deleted the blinds from the Tuiss integration, then deleted the BLE proxy from ESPHome.  I then re-did the BLE proxy first and then keyed in the blinds MAC addresses into the Tuiss integration.  May be worth putting this order in the instructions?

ValiantGB, 2024-07-04:
Thanks for the quick response!

closed 2024-07-04 by ValiantGB