pink88/Tuiss2HA #22: State of blinds

ellsa100, 2024-09-26:
Hi

I have a problem in that the Tuiss blinds status in Home Assistant only ever shows "**Opening**" or "**Closing**" and the Home Assistant status never shows them as "**Open**" or "**Close**", despite them actually being fully open or closed once operated through Home Assistant.

How can I fix this?

My other (non-Tuiss) blinds and curtains correctly show when they are "Open" or "Close" (as well as when they are "Opening" or "Closing".

Thank you!

pink88, 2024-09-26:
Hi Ellsa100,

Can you confirm that you are using the latest version 1.3.1 and the
model of blind you have?

Thanks,
Nigel

ellsa100, 2024-09-26:
Hi Nigel

I am using version 1.3.1. 

The blind is a Tuiss Electric SmartView for Roman Blinds from Blinds2go. Not sure how if that's enough for you to know the model number or if there is another place I can get more details about the model? I looked at the outside of the unit to see if there was anything else to help identify it, but all I could see without detaching from wall was battery identification details (ICR18650)

Thanks very much

pink88, 2024-09-26:
Two things:

   1. Are you using ESPhome bluetooth proxies to control the blind with
   Home Assistant? If not what?
   2. If you enable debug logging on the integration, then try to
   open/close the blind and share the log output with me.
      1. Go to Integrations > Tuiss > Enable debug logging
      2. Open/Close blind
      3. Go to Settings > System > Logs, scroll down and Load Full Logs,
      then scroll to the bottom and there should be some logs starting
      custom_components.tuiss2ha.hub

ellsa100, 2024-09-26:
Hi
1. Yes, using ESPHome bluetooth proxy (ESPHome 2023.11.6)
2. See below extract from debug
THANKS!

2024-09-26 15:39:32.504 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman E3:09:5D:57:BF:2B: TS3000: Attempting Connection to blind. Rety count: 0
2024-09-26 15:39:33.554 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman: Connected to blind
2024-09-26 15:39:33.747 DEBUG (MainThread) [custom_components.tuiss2ha.hub] Connected. Current Position: 0. Current Moving: -1
2024-09-26 15:39:33.749 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman: Attempting to set position to: 100
2024-09-26 15:39:33.778 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:33.778 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman: Sending the command
2024-09-26 15:39:34.953 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:35.955 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:36.956 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:37.958 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:38.960 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:38.965 INFO (MainThread) [homeassistant.components.automation.lights_turn_off_top_after_no_motion_for_1hr] Top floor Hall - Lights - motion auto: Running automation actions
2024-09-26 15:39:39.962 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:40.964 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:41.965 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:42.967 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:43.969 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:44.985 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:45.987 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:46.988 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:47.989 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:48.873 ERROR (MainThread) [homeassistant.components.emulated_hue.hue_api] Entity not exposed: 1
2024-09-26 15:39:48.990 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:49.991 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:50.996 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:39:51.997 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is False
2024-09-26 15:40:17.491 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman E3:09:5D:57:BF:2B: TS3000: Attempting Connection to blind. Rety count: 0
2024-09-26 15:40:18.506 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman: Connected to blind
2024-09-26 15:40:18.710 DEBUG (MainThread) [custom_components.tuiss2ha.hub] Connected. Current Position: 0. Current Moving: 1
2024-09-26 15:40:18.712 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman: Attempting to set position to: 0
2024-09-26 15:40:18.746 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:40:18.746 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman: Sending the command
2024-09-26 15:40:19.943 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:40:20.945 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:40:21.954 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:40:22.956 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:40:23.957 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:40:24.959 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:40:25.961 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
[... cut ...]
2024-09-26 15:41:26.823 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:41:27.824 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:41:28.826 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is True
2024-09-26 15:41:29.828 DEBUG (MainThread) [custom_components.tuiss2ha.hub] JacksRoman (E3:09:5D:57:BF:2B: TS3000) connected state is False

pink88, 2024-09-26:
That all looks correct to me. If you move the blind from Open/Closed to
50%, does that correctly show the Closing/Opening/Close/Open state?
I'm wondering if it's disconnecting part way through the process of opening
or closing, potentially if you have a particularly long blind. I do see
that occasionally with my own blinds, but it still 95% of the time will
correctly show the state.

ellsa100, 2024-09-26:
So in a couple of tests I did - if I stop midway after opening it seems to
think I am closed. Likewise if I stop midway through closing it seems to
think I am then open. But if it runs to the end it doesn’t know the status
and gets stuck on opening / closing.

I don’t think I have particularly long blinds, but it does take about 35
seconds to fully open or close if that means anything ?

pink88, 2024-09-26:
Okay. It sounds like a bug with it disconnecting during the full open and
close. I'll take a look at that and hopefully push a fix of sorts
this weekend. The blinds part open/closed logic is if its 25% or more OPEN
then state = OPEN, otherwise state = CLOSED. If its opening or closing,
then the state should be OPENING or CLOSING respectively.

I wouldn't expect this to be an issue every time though. I have a blind
that takes a similarly long time and the status is accurate most of the
time.

In any case, watch this space.

ellsa100, 2024-09-27:
oh great, thanks v much and look forward to the update.

pink88, 2024-10-05:
Version 1.3.2 has improvements in this area. Please update and let me know if things are improved for you. I've been running it locally for a week and not noticed any "stuck" states anymore.

ellsa100, 2024-10-05:
Thank you!

I’ve just updated and opened and closed a few times and seems to work :).

I’ll monitor the status over next few days and let you know if the issue
comes back, but hopefully it has been fixed for good.

Thank you so much.

closed 2024-11-24 by pink88