pink88/Tuiss2HA #23: Cover set action, nonetype object no attribute 'details'

StevieG10001, 2024-10-14:
Hello,

I've tried to connect 4 blinds to go (3 roman and 1 roller).   I.get the attached message, when I try to use the blinds.The blinds work fine with the remote control. I'm afraid I'm rapidly trying to learn Python and YAML, so this is well beyond me!

[Screenshot_20241012_023303_Home Assistant](https://github.com/user-attachments/assets/659cc2db-d1f1-444d-a19f-0c8b4cd997b7)

Have you come across this issue?  I've tried re installing blinds (and I have the MAC numbers).

Thanks in advance.

Steve

pink88, 2024-10-15:
Hi Steve,

This error is due to home assistant being unable to connect to the blind
via BLE, likely because the blind is too far from the bluetooth transmitter
or an unsupported bluetooth transmitter is being used. This integration is
only tested and designed to work using ESPHome and Bluetooth Proxies.

Can you confirm how your bluetooth set up is configured and if it is using
ESPHome, can you move the esp32 closer to the blind and retry?

Thanks,
Nigel

StevieG10001, 2024-10-15:
Hi Nigel,Thank you for the fantastically swift response!I'm now going to show my very novice credentials. I have set up ESPHome on my Home Assistant. I was hoping that the Home Assistant Connect thst's attached would reach the nearest blinds, which is probably 15 foot away (through a stud wall).I've got a ESP32 Bluetooth proxy on it's way from Pi Hut. Perhaps I get that attached and then try again.RegardsSteve

pink88, 2024-10-18:
These errors are due to Home Assistant being unable to connect to the blind. Can you try three things:

1. Can you confirm that you have the relevant bluetooth proxy set up in your M5's configuration file? Should look something like:

>esp32_ble_tracker:
>  scan_parameters:
>  interval: 1100ms
>  window: 1100ms
>  active: true
>bluetooth_proxy: 
>  active: true
>  cache_services: true`

2. If yes, when you open the log for the device can you see some lines like:

>[13:00:47][C][bluetooth_proxy:091]: Bluetooth Proxy:
>[13:00:47][C][bluetooth_proxy:092]:   Active: YES
>[13:00:47][C][bluetooth_proxy:093]:   Connections: 3
>[13:00:47][C][bluetooth_proxy:094]:   Raw advertisements: YES
>[13:00:47][C][safe_mode.button:024]: Safe Mode Button 'Safe Mode Boot'
>[13:00:47][C][safe_mode.button:024]:   Icon: 'mdi:restart-alert'
>[13:00:47][C][factory_reset.button:011]: Factory Reset Button 'Factory reset'
>[13:00:47][C][factory_reset.button:011]:   Icon: 'mdi:restart-alert'
>[13:00:47][C][esp32_ble:391]: ESP32 BLE:
>[13:00:47][C][esp32_ble:393]:   MAC address: 40:22:D8:78:71:06
>[13:00:47][C][esp32_ble:394]:   IO Capability: none
>[13:00:47][C][esp32_ble_tracker:655]: BLE Tracker:
>[13:00:47][C][esp32_ble_tracker:656]:   Scan Duration: 300 s
>[13:00:47][C][esp32_ble_tracker:657]:   Scan Interval: 1100.0 ms
>[13:00:47][C][esp32_ble_tracker:658]:   Scan Window: 1100.0 ms
>[13:00:47][C][esp32_ble_tracker:659]:   Scan Type: ACTIVE
>[13:00:47][C][esp32_ble_tracker:660]:   Continuous Scanning: True

3. Enable Debug Logging on the Tuiss integration, then share the Home Assistant logs that are generated when trying to use the blind.

StevieG10001, 2024-10-19:
Will do.  Thanks Andrew.S

phiebs, 2024-10-30:
Just hijacking this thread - I received (today!) a single tuiss/blinds2go blind. Have set it up in the tuiss app etc and am now trying to get HA integration using this awesome repo, with a view to getting a load more around the house if successful.
I'm making extensive use of Shelly's with BLE throughout the house and this works great for presence tracking etc but am getting the same error as the OP here, presuming as the Shelly isn't supported by this repo? Question is, I can put ESPHome on the Shelly's and I'm sure that will work, but is there something I can do to help contribute and add the needed code to enable the Tuiss's to work over Shelly's BLE? Selfishly this will avoid putting custom firmware on a number of Shelly's for me but could benefit others if it works? Thanks!

phiebs, 2024-10-30:
So I tried with the blind well within range (1 metre) of a vanilla ESPHome with BT enabled (confirmed by seeing presence updates of BLE devices) and the integration is exhibiting the same behaviour as described above by the OP.
Relevant lines from debug log:
`2024-10-30 14:57:56.637 DEBUG (MainThread) [custom_components.tuiss2ha.hub] Unable to find device En-Suite Blind, attempting rediscovery
2024-10-30 14:57:56.637 DEBUG (MainThread) [custom_components.tuiss2ha.hub] Unable to find device En-Suite Blind, attempting rediscovery
2024-10-30 14:57:56.638 DEBUG (MainThread) [custom_components.tuiss2ha.hub] Unable to find device En-Suite Blind, attempting rediscovery
2024-10-30 14:57:56.638 DEBUG (MainThread) [custom_components.tuiss2ha.hub] Unable to find device En-Suite Blind, attempting rediscovery
2024-10-30 14:57:56.638 DEBUG (MainThread) [custom_components.tuiss2ha.hub] Cannot find the device En-Suite Blind. Check your bluetooth adapters and proxies
2024-10-30 14:57:56.638 DEBUG (MainThread) [custom_components.tuiss2ha.hub] En-Suite Blind None: Attempting Connection to blind. Rety count: 0
2024-10-30 14:57:56.638 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140049825959216] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2761, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2804, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 996, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1068, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/cover.py", line 147, in async_open_cover
    await self.async_move_cover(1,0)
  File "/config/custom_components/tuiss2ha/cover.py", line 165, in async_move_cover
    await self._blind.attempt_connection()
  File "/config/custom_components/tuiss2ha/hub.py", line 116, in attempt_connection
    await self.blind_connect()
  File "/config/custom_components/tuiss2ha/hub.py", line 125, in blind_connect
    client: BleakClientWithServiceCache = await establish_connection(
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 343, in establish_connection
    if IS_LINUX and (devices := await get_connected_devices(device)):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/bluez.py", line 428, in get_connected_devices
    if not isinstance(device.details, dict) or "path" not in device.details:
                      ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'details'
2024-10-30 14:57:57.808 DEBUG (MainThread) [custom_components.tuiss2ha.hub] En-Suite Blind: Attempting to stop the blind.
2024-10-30 14:57:57.808 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140049825959216] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2761, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2804, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 996, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1068, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/cover.py", line 181, in async_stop_cover
    await self._blind.stop()
  File "/config/custom_components/tuiss2ha/hub.py", line 186, in stop
    _LOGGER.debug("%s: Stop failed. %s", self.name, self._client.is_connected)
                                                    ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'is_connected'
`
Which looks like it just can't find the blind but the blind is super super close. Is there anything else I can try to help diagnose?

phiebs, 2024-10-30:
It appears I need to RTFM, may be the same for OP. As soon as I set the ESPHome devices to use Active mode it works!
Now just to get the Shelly implementation of BLE working with them somehow!

pink88, 2024-10-30:
Good to hear! Re. the Shellys from what I can see they dont support
bluetooth control, just listening. See
https://community.home-assistant.io/t/shelly-plus-ble-no-devices-added-in-homeassistant/509199/21

phiebs, 2024-10-30:
Ahhhh that is unfortunate! Oh well I'm sure I can hide some ESP32's around the place! Thank you once again for a great integration :)

paddlemore, 2024-11-23:
I am having the same problem using an M5 Atom Lite as Bluetooth Proxy.  Please can you advise how I can check the M5 config file.  Can I do that from within Home Assistant or do I need to access the ESP32 via another means?  TIA

pink88, 2024-11-24:
Please refer to the esphome documentation for setting up esphome bluetooth proxy. Documentation can be found at esphome.io

closed 2024-11-24 by pink88