pink88/Tuiss2HA #27: 'None_Type' object has no attribute

robconway, 2025-01-04:
Thanks for creating this integration
I I have been running this using the raspberry pi BT no issue until it suddenly start throwing up the “none type” error
I have read through your closed issues and on reading them bought an ESP home, however having installed it and put it next to the blind it still doesn’t work.

how would you like me to work through the error and provide any log files to try and troubleshoot as I am now lost.

pink88, 2025-01-05:
If you enable debug logging for the integration, attempt to use the blind
and then share the logs that will help. Can you also share the
configuration for you esphome device?

robconway, 2025-01-05:
@pink88 I have no idea why but it suddenly started working. 
I cannot comment on how or why as I disconnected the ESP and blind in multiple ways and reconnected. Then just gave up. 
Anyway - thanks for the integration it is great!

closed 2025-01-05 by robconway