pink88/Tuiss2HA #28: Failed to perform the action cover/open_cover. 'NoneType' object has no attribute 'details'

JimCyberGoat, 2025-01-06:
HomeAssistant: 2025.1.0

Error: 
`Failed to perform the action cover/open_cover. 'NoneType' object has no attribute 'details'`

Log: 
`Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:245
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 18:40:51 (2 occurrences)
Last logged: 18:41:04

[547366043456] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2795, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2838, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1006, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1078, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/cover.py", line 195, in async_open_cover
    await self.async_move_cover(1,0)
  File "/config/custom_components/tuiss2ha/cover.py", line 214, in async_move_cover
    await self._blind.attempt_connection()
  File "/config/custom_components/tuiss2ha/hub.py", line 117, in attempt_connection
    await self.blind_connect()
  File "/config/custom_components/tuiss2ha/hub.py", line 126, in blind_connect
    client: BleakClientWithServiceCache = await establish_connection(
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/bleak_retry_connector/__init__.py", line 343, in establish_connection
    if IS_LINUX and (devices := await get_connected_devices(device)):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bleak_retry_connector/bluez.py", line 427, in get_connected_devices
    if not isinstance(device.details, dict) or "path" not in device.details:
                      ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'details'`

Have attempted a reboot and adding additional blinds.

JimCyberGoat, 2025-01-06:
Deleting and re-adding the blinds seemed to do the trick! Thanks for the ace integration.

TomCahill, 2025-01-09:
I've been running into the same issue but haven't had chance to debug it. Deleting and re-adding the entity hasn't resolved it for me.

I had it working fine on the Pi3b+ but after upgrading to a Pi5 (Imported Settings). I started running into the issue.

Same error and log as above.

JimCyberGoat, 2025-01-10:
> I've been running into the same issue but haven't had chance to debug it. Deleting and re-adding the entity hasn't resolved it for me.
> 
> I had it working fine on the Pi3b+ but after upgrading to a Pi5 (Imported Settings). I started running into the issue.
> 
> Same error and log as above.

I did find second time round, the MAC address field is case senstive. Could that be whats tripping you up too?

TomCahill, 2025-01-10:
> I did find second time round, the MAC address field is case senstive. Could that be whats tripping you up too?

Thanks for the heads up, that could be an easy fix but I don't think the case sensitivity is catching me out. It's my understanding that the MAC addresses are hexadecimal and should be always be (A-F, 0-9), so that shouldn't be the issue in the way I've entered it.

It looks like I'm referencing the entity correctly, as it can read the battery sensor value. However, the bind operating features are the only thing that's borked.

I'll hopefully get time over the weekend to try and dump so more logs out.

pink88, 2025-01-11:
That's very strange behaviour Tom. Does the blind operate when using the
set_blind_position action? Are you using bluetooth proxies or RP5 inbuilt
bluetooth?

GaryDubz, 2025-01-11:
I am also getting this error msg. my blind is on the top floor of my town house with the bluetooth transmitter on the ground floor. do you think this would be the problem and how would i get round this - Sorry... Newbee to HA

seanmaskey, 2025-01-12:
I have added a second blind which is reporting the same message -'Failed to perform the action cover/stop_cover. 'NoneType' object has no attribute 'is_connected' 
The first continues to work normally.
Pi 5 Raspbian HAOS 14.1 HA Core 2024.13.3. 
bluetooth proxy on atom stack running esp
[debug file.txt](https://github.com/user-attachments/files/18388927/debug.file.txt)
can't decode the file, but there clearly an issue its picked up.

Thanks for developing this; its a  neat integration.

pink88, 2025-01-19:
Hi seanmaskey,

Your error looks to be with the bluetooth adapter, either interference or
the proximity to the blind. bleak_retry_connector is a build in library of
home assistants, so not something I can control.

I will release an update soon that has better handling for the MAC address
case issue and to better handling of errors in general.

seanmaskey, 2025-01-19:
Thank you. That’s helpful to understand; I’ll try it with and without a Bluetooth proxy as this blind is quite close to the Pi, unlike the first which is on an atom m5 stack esp home device. 

S

ravenslea, 2025-01-29:
Thank you for this. I had exactly the same issue. Removing and adding again in CAPS for the MAC address worked.  I also just wanted to say thanks for this awesome integration which I am using all the time and love 🗡

aszabogit, 2025-02-01:
I had the same issue via ESPHome Bluetooth Proxy over the last 24 hours soon after updating to 2025.1.4, (but not immediately so don't think that was the cause).
After uninstalling a blind and adding it back in, it still didn't make any difference.
I then went to the Bluetooth Proxy configration and Installed ESPHome again via OTA and all the blinds started working again.

I've been using the integration for over a year and really appreciate it.

closed 2025-09-09 by github-actions[bot]