pink88/Tuiss2HA #31: Lot of connection issues after upgrading to 1.4.1

Fredzxda, 2025-01-27:
Hello,

I use proxy BLE to control my blind2go. After upgrading to 1.4.1 It is very difficult to connect to the blind.
It is the case for command send from Home Assistant and the same from the tuiss app on my smartphone.
Going back to 1.4.0 solve the problem.
I assume there is a lot of attempt which use all the bandwith of BLE or something like that.

Hope this can help.

pink88, 2025-01-27:
Thanks for flagging, can you confirm back if it's still an issue on 1.4.0
after a couple of days please.

Fredzxda, 2025-01-29:
The problem is still there with 1.4.0 version. Howerver, it seems to append less frequently.
For better understanding:

- I use 4 NSPanel with Blackymas blueprint [NSPanel_HA_Blueprint](https://github.com/Blackymas/NSPanel_HA_Blueprint)
- I enable ble proxy module integrated as an option.
- It work very well with my OralB toothbrush for a while.
- Looking at [NSPanel_HA_Blueprint issue tracker](https://github.com/Blackymas/NSPanel_HA_Blueprint/issues/1111) it seems they are usinf IDF version instead of arduino version for ROM and RAM constraints.

For the following days, i will try the different configuration to identify if it solve the problem:
1. Disable proxy module and keep it only for one NSPanel
2. Disable all proxy module and use a 8266 ESP with official proxy module.

I let you know the result by this weekend.

Fredzxda, 2025-02-01:
Here some updates:
The try to disable NSPanel proxy and keep only one was not succesfull. After 1 day, the connection problem reappear.
I flash an ESP32 card with latest proxy module and it work for more than 2 days. I am still in 1.4.0 and will move to 1.4.1 in a couple of day if the connection is still stable.
During this time, I will have a look at NSPanel proxy module code to try to identify the problem.

Fredzxda, 2025-02-09:
After several days with stable connection in 1.4.0, I switch today to 1.4.1. I let you know if the problem is solved next weekend.

Fredzxda, 2025-03-01:
No more problem even with 1.4.1. The issue can be closed.

closed 2025-03-01 by Fredzxda