pink88/Tuiss2HA #34: TS5200 Support - Blinds2Go change

rglss, 2025-03-06:
Hi,

It seems that Blinds2Go have changed the motors they ship with their electric blinds. I have some TS3000 devices that work a treat, however my latest blinds that have arrived today (6/3/25) have a TS5200 device (as reported by the Bluetooth monitor).

So far, these add succesfully to the integration, and the battery status is reported without issue. However the blinds don't respond to any commands.

An additional change is they also now have a button on the motor to initialise pairing.

I'll do some further investigation, any tips or guidence on the PCAPs required to investigate further is much appriciated.

pink88, 2025-03-06:
Thanks for sharing. Interesting to hear. I'll share my analysis and reverse
engineering from the TS3000 when I can tomorrow or the weekend and
hopefully they've only changed slightly the logic.

rglss, 2025-03-08:
I've tested the above with a second new TS5200 blind, and can confirm after adding to the Tuiss app it functions correctly with the integration. Will submit a PR with updated instructions in due course.

rglss, 2025-03-10:
Blinds have been tested and functioning fine over the last 3 days with the above. Submitted PR #35 to add instructions.

closed 2025-03-10 by rglss