pink88/Tuiss2HA #47: already_in_progress when trying to add Tuiss Smartview integration

xydas7, 2025-06-11:
Hello team,

I have setup HACS and the Smartview app on iOS (and everything works via the official app).

When I try to add the Tuiss Smartview integration in Home Assistant, after adding the correct mac address and giving it a name, I click confirm and get

"already_in_progress" and nothing else (apart from a close button).

How can I add the Smartview integration? 

Thanks

xydas7, 2025-06-16:
> pink88
>  left a comment
> (pink88/Tuiss2HA#47)
> I've not seen this error before, can you share any information from the logs?

I get the following screen…not really an error, but doesn’t let me do anything or add the integration in Home Assistant (this comes up on Home Assistant > Settings > Devices & Services > Add Integration > Tuiss Smartview and after I enter the correct MAC address and a name for the blind. The blind is already added and synced to the Tuiss Smartview App and works fine there).

Any suggestions?

ll￼

pink88, 2025-06-18:
Sorry, doesnt look like any image attached. If you can do the following:

1. Remove the blind from the integration
2. Click "Enable debug logging"
3. Add the blind
4. Click "Disable debug logging"

Share the file that was produced.

willbeeching, 2025-06-25:
Also having the same issue. This is my first blind I'm setting up and believe it's the latest one from Blinds2Go. Nothing in the logs and can't enabled debugging as I don't have any other blinds.

[image]

xydas7, 2025-06-26:
> pink88
>  left a comment
> (pink88/Tuiss2HA#47)
> Sorry, doesnt look like any image attached. If you can do the following:
>
> Remove the blind from the integration
> Click "Enable debug logging"
> Add the blind
> Click "Disable debug logging"
> Share the file that was produced.

I haven’t been able to complete the first step of adding the Tuiss2HA integration on Home Assistant, so I can’t follow your steps.

I followed Settings > Devices & Services > Add Integration > Tuiss Smartview and during that step I get one of the following:

1. “Already_in_progress”
or
2. Cannot connect

I can’t enable debug as the actual integration card is not added to my list (i.e. doesn’t appear under “Devices & services"

￼￼

pink88, 2025-06-26:
I tried to replicate on my system and haven't had any luck. Could you
remove from HACs and then download again?
Are any devices discovered automatically when you've installed the
integration?

xydas7, 2025-06-26:
The tuiss2ha is nowhere to be seen, I’ve never been able to install it and hence there is nothing for me to remove.

[image0.png]

Best,
Alexander Xydas

pink88, 2025-06-26:
Can you open HACS, search tuiss2ha, click ... and then remove. Restart and
then redownload from HACS and restart?

It should then auto discover any blinds if they are in range of Bluetooth.
If none appear, do you know what model of blind they are? Is should be TS
followed by 4 numbers.

xydas7, 2025-06-27:
Thanks for prompt response.

(Sorry newbie here) 
I can see Tuiss Smartview in 2 parts of my Home Assistant App.

1. Under the 3 lines (burger icon) > HACS - see screenshot - I have downloaded but taken no other action
2. Settings > Devices & Services > Add integration > Tuiss Smartview > Add your Tuiss Smartblind - this is what I tried in the past and got errors mentioned before.

My questions:
a) Which route is the correct one to follow? 
b) What do you mean will auto discover any blinds, where do I got to see this? 
c) where do I find my blind model (I only have the MAC and a QR code). My blinds can be controlled from a remote control and the iOS Tuiss Smartview app (assume it uses short range bluetooth, although the device does not show up in iPhone Settings > bluetooth). I purchased them from blinds2go (this one here <https://www.blinds-2go.co.uk/roller-blinds/52713/electric-cadiz-blackout-champagne.html?srsltid=AfmBOopkpPfRouaPQL1VI16nD68R7OVzGRWUK1kw3jVbiQRu8sGyKwX1>)

￼

pink88, 2025-06-28:
No Problem.

a) Which route is the correct one to follow?
Can you go through HACS. Under the three dots there should be an option to
Remove? Can you do that, restart Home Assistant. Redownload the integration
from HACS, restart. When you do this, you shouldnt need to add the
integration from Settings > Devices > Integrations, instead it should
autodiscover your blinds2go blind if its in range of bluetooth after a
couple of minutes. If it doesn't appear, try two things:

   1. If you can move your bluetooth adapter/proxy closer to the blind and
   see if it connects
   2. Enter the blind information manually by adding the integration
   through Settings > Devices > Integrations

b) What do you mean will auto discover any blinds, where do I got to see
this?
They should appear at the top of the Integrations in Settings > Devices >
Integrations
c) where do I find my blind model (I only have the MAC and a QR code). My
blinds can be controlled from a remote control and the iOS Tuiss Smartview
app (assume it uses short range bluetooth, although the device does not
show up in iPhone Settings > bluetooth). I purchased them from blinds2go
(this one here <
https://www.blinds-2go.co.uk/roller-blinds/52713/electric-cadiz-blackout-champagne.html?srsltid=AfmBOopkpPfRouaPQL1VI16nD68R7OVzGRWUK1kw3jVbiQRu8sGyKwX1
>)
Don't worry about this, I wasn't sure if the information was available
elsewhere

Thanks

exceladdict, 2025-08-30:
> [@exceladdict](https://github.com/exceladdict) sorry about this, pretty major bug. It's been fixed now in 1.9.5 [#70](https://github.com/pink88/Tuiss2HA/pull/70). You should just be able to use the auto discovered entry. Please let me know if the issue persists.

Have added many blinds now and all working well. Will look at using another Poe Bluetooth proxy for the final set of blinds soon.

Thanks so much for your amazing work on this @pink88!

gazes01logic, 2025-08-30:
I’m having this same issue but running 1.9.6 @exceladdict - are you running 1.9.5 or on the latest?

Will try reinstalling everything to see if that helps as well.

closed 2025-08-31 by pink88