pink88/Tuiss2HA #48: Tuiss Smartview spends a long time initialising

jgeorge1983, 2025-06-12:
So I dont really know where the issue lies here,  but wheneve i reboot homeassistant the thing that always takes the longest is Tuiss Smartview initialising.

often i go into my integrations and it is still initialising.

I am running homeassisntant(2025.6.0) via docker on a raspberry pi and have a bluetoothproxy around 1 meter from my smart blind.
running version 1.7.1 from HACS

If i use the official app it works no problem, just the integration doesnt seem to start most of the time.

Happy to provide anu logs/triage you think would be helpful.

Thanks,.

pink88, 2025-06-13:
I've released a prerelease which makes the startup checks optional (and off by default), which speeds up initialisaton significantly. Let me know if this improves things?

jgeorge1983, 2025-06-13:
installed and rebooted, much quicker.  thank you.  I'll wait a few hours then test the blind opening/closing.  thanks

jgeorge1983, 2025-06-13:
I tested, and got the error:
Failed to perform the action cover/close_cover. '<=' not supported between instances of 'int' and 'NoneType'

so i deleted the blind, it auto discovered it to readd and i get "Unknown error occurred" upon adding.

pink88, 2025-06-13:
I've deleted the old 1.8.0-beta1 and republished it with the fix, if you can redownload the version or rollback to 1.7.1 and then reupdate.
There was a bug in the way the amount of retry attempts was set. In my testing both the scenarios above are working now.

pink88, 2025-06-13:
> I just rolledback, to 1.7.1 and then installed 1.8.0-beta1, but i still
> get the Unknown Error on adding the blind.
>
> It is found automatically. just wont add

Odd. Can you enable debugging, add the blind and share the results again
please.

it moving millimeters was a connection thing, but shouldnt happen anymore.

jgeorge1983, 2025-06-13:
This is the error from the logs, do you need more?
```
Logger: aiohttp.server
Source: /usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py:481
First occurred: 12:02:38 (3 occurrences)
Last logged: 12:10:25

Error handling request from 192.168.4.205
Traceback (most recent call last):
  File "/config/custom_components/tuiss2ha/config_flow.py", line 137, in validate_input
    await hub.blinds[0].get_blind_position()
  File "/config/custom_components/tuiss2ha/hub.py", line 258, in get_blind_position
    await self.get_from_blind(command, self.position_callback)
  File "/config/custom_components/tuiss2ha/hub.py", line 238, in get_from_blind
    await self.attempt_connection()
  File "/config/custom_components/tuiss2ha/hub.py", line 109, in attempt_connection
    while retry_count <= self._restart_attempts:
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '<=' not supported between instances of 'int' and 'NoneType'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 510, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 41, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 83, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 227, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 336, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 383, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 487, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/config_flow.py", line 111, in async_step_confirm
    _title = await validate_input(self.hass, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/config_flow.py", line 140, in validate_input
    raise CannotConnect()
custom_components.tuiss2ha.config_flow.CannotConnect
```

pink88, 2025-06-13:
Thanks for this and your help. Looks like the last version didnt pick up the code changes I'd made for whatever reason. Have released 1.8.0-beta2.

jgeorge1983, 2025-06-13:
got it installed, blind added.  cant test it yet due to a sleeping baby.

although my bt proxy is only 1-2 meters away, I think it may be travelling through a steel lintel/beam as the blind is on the landing and the proxy is in the loft directly above.  I was checking the new bt features in HA and it has -85 rssi, deffo shoudlnt be that poor being so close.

smithey253, 2025-06-22:
> got it installed, blind added. cant test it yet due to a sleeping baby.
>
> although my bt proxy is only 1-2 meters away, I think it may be travelling through a steel lintel/beam as the blind is on the [...]

Jumping in on this, I also have slow response from the blind after pressing a button to raise/ lower the blind. My bluetooth proxy is only a metre or so from the blind with -77 RSSI. I know some ESP32s Bluetooth is pretty directional so I'm going to try rotating my ESP32 to see if I get any improvements.

jgeorge1983, 2025-06-23:
Really?! I didn't realise the direction mattered.  Can you update on the best direction when you find it please??

closed 2025-06-26 by pink88