pink88/Tuiss2HA #54: Double Roller Blind Support

CodyT, 2025-07-02:
I don't currently own any of these blinds and I couldn't find anywhere to discuss these apart from here on GitHub issues, apologies if there is somewhere better to post this question:

Does this integration work with double roller blinds? For example: https://www.blinds-2go.co.uk/roller-blinds/48081/electric-double-roller-atom-pure-white.html 

I'd love to go ahead and purchase a bunch of these for the double versions for bedrooms and single versions for living spaces.

JeffyGeoff, 2025-07-09:
Double rollers are just 2 roller blinds in the same double brackets so can be set up on the integration as 2 blinds.

CodyT, 2025-07-10:
Thank you @AndyCoo71 @JeffyGeoff I appreciate the response, I spoke to their customer support over the past days and they confirmed that their double rollers are 2 individual rollers. I'll go ahead and close the issue.

closed 2025-07-10 by CodyT