pink88/Tuiss2HA #65: TS5200 - no connection after a few hours

Mountain-wg, 2025-08-12:
I have a strange problem with one of my blinds. The older ones (5) are equipped with TS3000, are working as expected.

The TS5200 one acts strange. I use the Home Assistant Cloud Connection to Alexa where an Routine runs the blind 4 times a day.
Everything runs great until it stops after 1-2 days. No connection from Alexa, no connection directly from HA and also no connection from the Tuiss app: "unable to connect".
To solve this issue I restarted the blind manually by pressing the button at the back of the motor. After 1-2 days the same problem occurs.
It was clear to me that the blind motor/electronics were defect and I called Tuiss to exchange the blind. Got a new blind - same issue!

Finally I found out the other day, that after I restarted HA, the blind is accessible again - from HA and from the Tuiss app. So there seems not to be a problem with the motor or electronics, but with the HA Tuiss service. 

How can I help to fix the problem?

Thanks in advance

Mountain-wg, 2025-08-24:
So, this morning the routine was not able to move any blind (new and old motors).
HA was not able to connect to any motor/blind.
Tuiss app was not able to connect either.
Only a reboot of the HA system was able to get in contact with the blinds again. Reload of config was not enough, which was before.

As requested my detail log from the tuiss integration. Full log, no cut outs to see that they were fine in the beginning and ended up with no connection:

[home-assistant_tuiss2ha_2025-08-24T06-24-53.751Z.log](https://github.com/user-attachments/files/21955643/home-assistant_tuiss2ha_2025-08-24T06-24-53.751Z.log)

For the Bluetooth connection, I use an ESP32 with ESPHome Bluetooth Proxy with Firmware: 2025.8.0 (Aug 20 2025, 21:41:43). 

The setup / command chain looks like this:
Human Voice --> Alexa Show --> Home Assistant Cloud --> lokal HA --> Tuiss Integration --> Bluetooth Proxy --> Blind

pink88, 2025-08-24:
Please see if 1.9.6 has resolved the main issue with losing connection after some time. If the blind does lose connection part way through a command, it should timeout (might take 2-3 minutes) at which point the position and state of the blind will update in the UI. 

If you notice the position of the blind updating very slowly, move the blind ~25% and it should reset the logic .

I'm not sure that this will completely resolve the problem, so if it reoccurs, please share logs and I'll have another look.

HumanSkunk, 2025-08-25:
Installed 1.9.6 last night remotely and the blind opened fine this morning. When a second automation ran to adjust the height it just displayed the "opening state". Checking the logs and this is in the logs.

`2025-08-25 07:00:02.162 WARNING (MainThread) [custom_components.tuiss2ha.cover] Bathroom Blind: Timeout waiting for blind to stop
2025-08-25 07:00:03.283 WARNING (MainThread) [custom_components.tuiss2ha.hub] Bathroom Blind: Error disconnecting: Characteristic 00010304-0405-0607-0809-0a0b0c0d1910 was not found!`

Now back home It does still respond to commands. When I updated last night it switched off debug Logging off so I have turned it back on see if any new logs are generated.

Mountain-wg, 2025-08-25:
Good news from my side: 4 routines run without any problem.
Something interesting happed with one of the older blinds: HA had the job to close the blind and showed closed, but the blind was only 83% closed. The tuiss app showed the real status. After a complete opening and closing of the blind, it seems fixed. Maybe not related tot 1.9.6 but worth mentioning. 
Will monitor the blinds for the next days from a remote location. Debug log is on.

Tau512, 2025-08-26:
looks like both blinds have gone offline for me now.
HA shows them in their correct ending position, but stuck in the "opening/closing" state. 

log (hopefully sanitised):
[home-assistant_tuiss2ha_2025-08-26T17-33-41.570Z.log](https://github.com/user-attachments/files/21992755/home-assistant_tuiss2ha_2025-08-26T17-33-41.570Z.log)

plenty of BT slots on my ESP BT proxies; iirc i've only got these 2 blinds using bluetooth so the 2x ESP32 proxies are pretty much exclusively for them. both ESP's should be in range of both blinds too.

Mountain-wg, 2025-08-26:
2 Days without any problem - thanks very much for the great work!
I stopped my debug logging, but I have an eye on it and would report, if I notice new (old) problems

HumanSkunk, 2025-08-29:
Had an issue for the first time again in a while this morning typically after I turned off debug logging yesterday as all things seem fine.

This morning the status was on opening even though the blind was fully open. What differed this time was that I could still control it in HA. After I moved it down a bit I tried to move it back up and I got connection issues was unresponsive to both HA and wouldn’t connect by the Tuiss app. It doesn’t seem to like multiple movements in succession, is that a limitation in how the Bluetooth proxy operates?

After a few minutes it was controllable again so it doesn’t completely lock itself out so that timeout seems to work.

If I just left it with the usual automations just simply opening and closing I don’t think you would ever encounter an issue now. It just seems to be finicky you send a few commands in succession. I have turned debug back on and will keep it on this time.

The logs it produces are the same as what I captured before when it goes into its lockdown state. I don’t have any of it actually initiating the moves.

```
2025-08-29 08:05:48.578 WARNING (MainThread) [custom_components.tuiss2ha.cover] New Bathroom Blind: Timeout waiting for blind to stop
2025-08-29 08:05:50.449 WARNING (MainThread) [custom_components.tuiss2ha.hub] New Bathroom Blind: Error disconnecting: Characteristic 00010304-0405-0607-0809-0a0b0c0d1910 was not found!
```

Mountain-wg, 2025-08-29:
FYI - what I experienced is that the shades are not moving simultaneously. I found this explanation:

> An [ESPHome]Bluetooth proxy supports a maximum of three active connections simultaneously. Devices that only passively broadcast data, such as many sensor devices, do not count toward this limit, as they don't establish a continuous, active connection. You can overcome the limit by using multiple Bluetooth proxies, as you can place them in different areas to cover more devices.
[https://esphome.io/components/bluetooth_proxy/#:~:text=The%20Bluetooth%20proxy%20of%20ESPHome,limited%20to%20a%20specific%20number.](url)

I will try to install a 2nd Bluetooth proxy in the next weeks to see if this changes

Tau512, 2025-08-29:
Very true about the 3 devices with ESP BT proxy. Thankfully i stumbled into that when i deployed these Tuiss blinds & required my first BT connected devices (so went with ESP BT Proxy). They're still the only bluetooth devices deployed.

2 blinds, 2x ESP BT proxies (1 in each room with the blinds) so plenty of capacity & signal strength

I've had about 36hours of good functionally i thought i got my last comment wrong. But alas, problems with the 'existing connection not completed' message started again yesterday, and again today. Not sure if it's a specific problem or a coincidence, but I'm only seeing the issue with the blinds when fully up. not had them be a problem with them in the 50-100% closed range, with an "opening" state.
A restart of HA seems to fix the issue so i guess i could probably automate a fix for that if either blind is in an "-ing" state for >1hour then it's probably bugged out.

norfolkmustard, 2025-08-29:
Liking the (new I think?) verbose messages in automations when they fail - I had this through the day today on 2 x TS3000

> Error: Lounge Front Roman is locked, please wait for currrent command to complete and then try again.
> Error: Kitchen Front Roman is locked, please wait for currrent command to complete and then try again.

It looks like these were locked for 13 hours today.

As soon as I reloaded the integration they were responsive again. Other blinds (TS3000, TS2600, TS5200) working ok before and after too.

HumanSkunk, 2025-08-30:
Same issue again tonight but this time it completely locked up. It does seem to be something to do with the ESPHome BT Proxy as soon as I temporarily disabled it within the ESPHome Integration the Tuiss app was able to control the blind again.

When i enabled the BT Proxy again I was still unable to control it from within HA getting a load of ‘connecting’ messaged in the BT Proxy logs. I think initially it was becuase i still had the Tuiss app open. One closed it still took a minute or so and then it was able to connect and update the status.

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:317
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 29 August 2025 at 08:03:06 (20 occurrences)
Last logged: 23:35:29

[546758968160] Error during service call to cover.open_cover: New Bathroom Blind is locked, please wait for currrent command to complete and then try again.
[546508103552] Error during service call to cover.set_cover_position: New Bathroom Blind is locked, please wait for currrent command to complete and then try again.
[546675128288] Error during service call to cover.close_cover: New Bathroom Blind is locked, please wait for currrent command to complete and then try again.
[546676442272] Error during service call to cover.set_cover_position: New Bathroom Blind is locked, please wait for currrent command to complete and then try again.
[546494687456] Error during service call to cover.open_cover: New Bathroom Blind is locked, please wait for currrent command to complete and then try again.
```

Here are logs off the proxy

```
23:42:33][W][bluetooth_proxy:194]: [0] [XX:XX:XX:XX:XX:XX] Connection request while pending disconnect, cancelling pending disconnect
[23:42:33][W][bluetooth_proxy:043]: [0] [XX:XX:XX:XX:XX:XX] Connection request ignored, state: CONNECTING
[23:42:40][D][esp32_ble_client:356]: [0] [XX:XX:XX:XX:XX:XX] ESP_GATTC_DISCONNECT_EVT, reason 0x100
[23:42:40][D][bluetooth_proxy.connection:380]: [0] [XX:XX:XX:XX:XX:XX] Disconnect, reason=0x100
[23:42:40][D][esp32_ble_client:196]: [0] [XX:XX:XX:XX:XX:XX] ESP_GATTC_OPEN_EVT
[23:42:40][D][esp32_ble_client:293]: [0] [XX:XX:XX:XX:XX:XX] ESP_GATTC_OPEN_EVT in IDLE state (status=133), ignoring
[23:42:40][D][esp32_ble_tracker:139]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[23:42:40][D][esp32_ble_tracker:867]: Setting coexistence preference to balanced.
[23:42:40][D][esp32_ble_tracker:219]: Starting scan, set scanner state to STARTING.
[23:42:44][I][bluetooth_proxy:048]: [0] [XX:XX:XX:XX:XX:XX] Connecting v3 with cache
[23:42:44][W][bluetooth_proxy:043]: [0] [XX:XX:XX:XX:XX:XX] Connection request ignored, state: DISCOVERED
[23:42:44][D][esp32_ble_tracker:139]: connecting: 0, discovered: 1, searching: 0, disconnecting: 0
[23:42:44][D][esp32_ble_tracker:821]: Stopping scan to make connection
[23:42:44][D][esp32_ble_tracker:829]: Promoting client to connect
[23:42:44][D][esp32_ble_tracker:863]: Setting coexistence to Bluetooth to make connection.
[23:42:44][I][esp32_ble_client:111]: [0] [XX:XX:XX:XX:XX:XX] 0x01 Connecting
[23:42:44][D][esp32_ble_client:209]: [0] [XX:XX:XX:XX:XX:XX] medium conn params
[23:42:44][D][esp32_ble_tracker:783]: Scan stop complete, set scanner state to IDLE.
[23:42:44][D][esp32_ble_tracker:139]: connecting: 1, discovered: 0, searching: 0, disconnecting: 0
[23:43:03][D][esp32_ble_client:356]: [0] [XX:XX:XX:XX:XX:XX] ESP_GATTC_DISCONNECT_EVT, reason 0x100
[23:43:03][D][bluetooth_proxy.connection:380]: [0] [XX:XX:XX:XX:XX:XX] Disconnect, reason=0x100
[23:43:03][D][esp32_ble_client:196]: [0] [XX:XX:XX:XX:XX:XX] ESP_GATTC_OPEN_EVT
[23:43:03][D][esp32_ble_client:293]: [0] [XX:XX:XX:XX:XX:XX] ESP_GATTC_OPEN_EVT in IDLE state (status=133), ignoring
[23:43:03][D][esp32_ble_tracker:139]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[23:43:03][D][esp32_ble_tracker:867]: Setting coexistence preference to balanced.
[23:43:03][D][esp32_ble_tracker:219]: Starting scan, set scanner state to STARTING.
```

Mountain-wg, 2025-08-31:
@HumanSkunk sad to hear the many problems you have, reminded me of my frustrational experience before.

@pink88: sill no problems on my side since the 1.9.6. - 7 days without any issue. As mentioned, I will try to upgrade to more than one BT Proxy the week after the next week to see if i can have a more synced rise of the shaded. But this is only the cherry of the icing. I'm so happy with your work, I would love to benefit you in any way (buymeacoffee, donate, amazon)

pink88, 2025-08-31:
@Mountain-wg  really glad to hear this is working well for you again. Been a tough one to try to pinpoint, but genuinely, learning and spending time trying to fix for you is reward enough, though thanks for the offer!

@HumanSkunk I'm looking for some solutions, I have a couple ideas, will hopefully push out a prerelease tomorrow for you to try out.

HumanSkunk, 2025-08-31:
@pink88 thanks for the update. I have now installed a mmwave sensor in the bathroom which is also now used as a BT proxy. Seems to be a bit more reliable now it's split between the two as the proxy in the kitchen is quite far away from the bathroom and had a low signal strength.

pink88, 2025-09-01:
@HumanSkunk  I've released prerelease 1.10.0-beta1. It includes some more tweaks to the connection and timeout logic. If you can give it a download and let me know if things are improved or that, it would be much appreciated. I've seen one timeout in my testing, but the state and position of the blind corrected itself after the timeout lifted (2 mins).

Tau512, 2025-09-01:
you can include me in the testing!
just updated to 1.10.0-beta and will see how it goes.  🤞 

for others; since it was the first time i've had to install a beta version via HACS and wasnt entirely clear:  HACS > Tuiss2HA > 3dot menu > Redownload. can select the pre-release from there

AndyCoo71, 2025-09-01:
I’m having the same problems with blinds stuck in Opening or Closing even after they’ve finished moving so would like to try the beta too.
But I couldn’t follow those install steps. Sorry, where do I start?

Tau512, 2025-09-01:
in HA, open HACS and search for the tuiss2HA app:
 [image]

click the 3 dots on the right, next to the addon (to the upper right corner dots) and a menu should appear. 

click 'redownload'

click 'need a different version', and select the beta from the list & download.
[image]

if you do not see the 1.10.0-beta1 listed, go back to the 3dot menu and select 'update information' and re-attempt the redownload steps. HACS periodically updates, but i'm not sure on the frequency.

HumanSkunk, 2025-09-01:
It has been behaving the last 24 hours, but I will update and see what happens.

Is there a limitation on running multiple successive movements through the integration? Is that what causes the potential locked state? Content with just having the automations opening and closing them at set times. And occasionally them responding to a request.

Tau512, 2025-09-02:
i think the beta1 release has improved things, as i've only got 1 (of 2) blinds offline this morning. Typically both would be unresponsive.

`Failed to perform the action cover/set_cover_position. Kitchen is locked, please wait for currrent command to complete and then try again.`
[home-assistant_tuiss2ha_2025-09-02T05-37-07.221Z.log](https://github.com/user-attachments/files/22087792/home-assistant_tuiss2ha_2025-09-02T05-37-07.221Z.log)

very minor thing i've just seen too - there's 3 R's in that "please wait for **currrent** command to complete"

HumanSkunk, 2025-09-06:
Things seem ok for me now. I am only raising and lowering once per day but blinds have remained responsive.

Tau512, 2025-09-06:
i've been meaning to check back in on this: seems its been 4days since my last message! Everything has been fine since then - reliable blind operation and no obvious issues with my automations or any error messages.

Tau512, 2025-09-26:
There's been a vast improvement in reliability with recent changes but I'm still seeing an infrequent connection failure on one of my two blinds. seems to happen on the first/last movement of the day but, I'm sure that's not the exact reason and just coincidence.
 
I see maybe 3 or 4 failures since my last post 3 weeks ago. The blind will move to it's requested position, but then HA gets stuck in opening/closing state requiring a restart of HA to work again as per the ongoing issue.

my blinds typically have 6 movements per day so does function fine majority of the time.

currently on the latest 1.10.0 stable version

NiblettS, 2025-09-26:
I have 5 blinds synced (1.10.0) to move together and I found all of my issues were actually with my esp32 configuration. The location is near a street with high traffic so the Bluetooth connectivity are busy most of the time. I doubled up on esp32 and now seem to very rarely get any issues.

Tau512, 2025-09-29:
i've been running multiple ESP32 proxies for many months so it's not specifically that as the fault.
however i have also been running through HA logs and cleaning up some old un-used plugins like BT location & triangulation. wondering if those were tying up connections but i think i've seen something along the lines of "max connections met" when there isn't enough connection slots.

hopefully i've cleaned up enough; will see how things go over the next week+ with only Tuiss2HA app running on bluetooth.

PR #75 (2025-09-01): Simultaneous action
- Adds a new action to allow connecting to multiple blinds simultaneously, so they operate at exactly the same time. #72 
- Additional rework to the connection logic to try to resolve #65 
- Remove blind orientation option flow as no longer needed