pink88/Tuiss2HA #72: Control two blinds at one time?

PatrikHAOS, 2025-08-24:
Hello!

I use this for ny two blinds and the idea is to over time exchange all blinds to TUISS (or similar), it seems that there is a feature in the official app that I can not understand how to do with this integration, and that is to control two (or more) blinds at the same time. In the official app this can be done.

Is there a possibility to add this functionality? I have tried to create a group in HAOS but that controls them still one by one, and not at the same time.

Thanks for a great integration!

pink88, 2025-08-27:
Good idea. I have this working locally now, so it will be a part of the next release planned in a week or so.

patrikjunno, 2025-08-30:
On this topic. It sounds like functionality is on its way but does it mean that the integration will seek and confirm connection with a selected group of devices before action? In my case I have four windows with roller-blinds next to each other. I have created a group in HA but when I for instance open the group they will not open in sync but in a totally random order and with different amount of delay. They eventually end up in the same position.

According to the connection-strength entity they have similar connection (between -85 and -89). What is considered a good enough connection by the way?

Appreciate all effort and if you want help testing please let me know. 

Thanks

patrikjunno, 2025-08-31:
Thanks for clarifying. The remote and app seems to be able to connect with four blinds to operate them simultaneously but you mean that there's a limitation in the HA integration? 

Would you mind explaining how and with what to set up two proxies? I understand that this is another topic but for the purpose of operating the blinds we might be able to consider it "on-topic" 

 I just have HA OS installed on a Lenovo tiny pc today with a Bluetooth/wifi-antenna.

Keep up the good work

NiblettS, 2025-09-05:
I have 5 blinds around a bay window and they are connecting via ESP32 device configured for 5 connections. 
The blinds sequentially is totally erratic so would like to try this out, and have installed 1.10.0-beta2 (https://github.com/pink88/Tuiss2HA/tree/1.10.0-beta2) , I am assuming holds the new code as well.

Got everything running but my yaml is very poor, to you have a pseudo script I can use as a template, and feedback?

pink88, 2025-09-06:
@NiblettS  could you share a couple things please.
1. Can you go into the Tuiss integration (Settings > Devices & Services > Tuiss > Click each blind) and let me know what the signal strengths are of the blinds in question
2. Can you let me know the traversal_speed of the blinds (press c > search developer tools > states > search the blind name)

In my testing, its only as fast as home assistant will manage the connections and seems about as fast as connecting to multiple blinds in the Tuiss app has been for me.

pink88, 2025-09-07:
That all looks about the same as my set up. I'm not sure what can be done, but I'll do some poking around.