pink88/Tuiss2HA #83: " Unknown error. Please try again. If you consistently get this error please raise an issue."

Zetascrub, 2025-09-29:
Long story short, I've been using this custom component for a few weeks, for the most part, works well. But in the last couple of days, it just barely works and today, not at all.

I've deleted the component and reinstalled it and now I can't add the blinds?

I'm getting the " Unknown error. Please try again. If you consistently get this error please raise an issue." error during the integration adding phase.

The log entry shows as follows

```
This error originated from a custom integration.

Logger: custom_components.tuiss2ha.config_flow
Source: custom_components/tuiss2ha/config_flow.py:93
integration: Tuiss Smartview (documentation, issues)
First occurred: 15:47:51 (5 occurrences)
Last logged: 16:01:00

Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/tuiss2ha/config_flow.py", line 93, in async_step_user
    _title = await validate_input(self.hass, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/tuiss2ha/config_flow.py", line 165, in validate_input
    await hub.blinds[0].get_blind_position()
  File "/config/custom_components/tuiss2ha/hub.py", line 349, in get_blind_position
    await self.get_from_blind(command, self.position_callback)
  File "/config/custom_components/tuiss2ha/hub.py", line 325, in get_from_blind
    await self.attempt_connection()
  File "/config/custom_components/tuiss2ha/hub.py", line 151, in attempt_connection
    raise DeviceNotFound(
        f"{self.name}: Cannot find the device. Check your bluetooth adapters and proxies"
    )
custom_components.tuiss2ha.const.DeviceNotFound: Kitchen Blind: Cannot find the device. Check your bluetooth adapters and proxies
```

I've checked that HA can see and use the bluetooth adaptor, I've used the official app and the blinds batteries are good. One of the blinds is 3 feet from the bluetooth adaptor so the distance can't be an issue.

Any idea?

squozeneu, 2025-10-02:
I have just received some blinds and the app works. I am running HA in a VM and have passed through the bluetooth adapter which HA sees. I am getting the same error and I'm sure it's something I'm doing wrong. Maybe I haven't told Tuiss how to use the bluetooth adapter? Not sure what I should be doing.

squozeneu, 2025-10-04:
@pink88 I’m going to assume it’s my Bluetooth on the HA machine and buy a Bluetooth proxy. Is there any specific one you recommend or can I get any generic ESP32 board?

pink88, 2025-10-04:
I just use standard esp32 dev boards, like the first results on Amazon or
Google, but any that are supported by esphome would work (and I think
that's pretty much all of them).

squozeneu, 2025-10-16:
I stopped getting the error when I installed a BT proxy. Thanks!